WU-BIMAC/MicroMetaApp-React #18: Disappearing Images after pressing enter at the last entered value

mathomH, 2019-06-06:
I found that back at tier 1, examples:
1st "Edit microscope" filled Generic then "ManufactorerSpec" and hit enter -> white screen
(notice the last tab has no typing entering there = "ManufactorerSpec" is the last thing where there is anything to type)
2nd added an objective filled all information and pressed enter after the last entry -> white screen

I can not attach the saved microscopes to let you inspect the errors, so I gonna email them

Akramhar, 2019-06-06:
Thanks Mathias, I received your email with the jsons.
Will test tomorrow and try to fix it

Akramhar, 2019-06-10:
Problem couldnt be reproduced in version 0.0.5, more investigation is required

Akramhar, 2019-11-26:
this is probably not relevant anymore, I'll close it and if the problem reappears we are going to reopen it

closed 2019-11-26 by Akramhar