WU-BIMAC/MicroMetaApp-React #30: Pump Child Element Ref error

strambc, 2019-06-11:
The Pump Child Element should refer to a different light source reference with respect to the one that it is a child of.
For example, Laser 1 has Pump which should refer to another light source (e.g. another external laser, or an arc lamp, etc.).
At present it refers to the same ID as the parent Laser element.

strambc, 2019-06-11:
When attempting to add as second Laser element, there is an issue with saving the element.
When fields are filled in, then one gets to the Pump tab and two things happen:

1. if no pump is added, confirm is not allowed
2. If one adds a pump, confirm is not allowed and all fields from previous tabs are cleared.

Akramhar, 2019-06-11:
This seem to be an error with the model.
This is how an EmissionFilter is represented inside the FilterSet.
` <xsd:element name="EmissionFilterRef" minOccurs="1" maxOccurs="1">
              <xsd:annotation>
                <xsd:appinfo><xsdfu xmlns=""><plural>EmissionFilters</plural><manytomany/></xsdfu></xsd:appinfo>
                <xsd:documentation>Description=This refers to an available emission filter, which is part of this filter set.</xsd:documentation>
                <xsd:documentation>Tier=2</xsd:documentation>
              </xsd:annotation>
            </xsd:element>`
while this is how the pump is represented inside Laser:
` <xsd:element ref="Pump" minOccurs="0" maxOccurs="1">
              <xsd:annotation>
                <xsd:documentation>Description=The Laser element may contain a Pump sub-element which refers to a LightSource used as a laser pump
                </xsd:documentation>
                <xsd:documentation>Tier=5</xsd:documentation>
              </xsd:annotation>
            </xsd:element>`

strambc, 2019-06-11:
OK I fixed the Pump, it is now behaving like "WavelengthRange"
Let me know if this works

Akramhar, 2019-06-12:
I think that conceptually this should work as "EmissionFilter" and not as "WavelengthRange" due to the fact that it's a reference to another element in the model (from what I gathered) and not a sub container for information.

strambc, 2019-06-12:
OK I spoke with Kevin.
Pump is going to be deprecated.

closed 2019-06-12 by strambc