WU-BIMAC/MicroMetaApp-React #32: Focusing vs. Stage

strambc, 2019-06-11:
Noticed there is a discrepancy between STAGE vs. FOCUSING in the model.
Currently Focusing extends Stage and all elements that belong to the Focusing group are of Focusing type.

However there are in fact two categories:

1) Stage (Mechanical and Piezo)
2) Focusing (Turrett and Piezo).

I would like to reorganize with nested groups (similar to Detector --> Camera).

Focusing group contains
1) ObjectiveTurrett
2) Piezolelectric Objective
3) PositioningDeviceGroup

PositioningDeviceGroup contains:
1) Mechanical Stage
2) Piezoelectric Stage

This way the user would not have to enter a lot of NOT APPLICABLE when filling in Focusing devices and only enter X and Y movement when applicable.

Would this create a lot of problems on your hand?

C

This way it would make more sense for users to fill in informations because

Akramhar, 2019-06-12:
Do these components share common attributes than we can extrapolate in a super class?
I understand the need for convenience but the detector / camera categories are created based on the model, because detector has a set of attributes that are shared by some components and camera has the same.
Doing it arbitrarily just to have components displaying in a different way it would feel, in my opinion, like ruining the model.

strambc, 2019-06-13:
It would not be arbitrary: it would be the separation of STAGE vs FOCUSING-DEVICE.
FOCUSING-DEVICE ONLY MOVE up and down
STAGE potentially moves in all direction (including up and down).
In this sense, STAGE IS-A FOCUSING-DEVICE.
Here is what I propose:

[image]

Akramhar, 2019-06-17:
Ok, needs to be modified in the model and tested out

strambc, 2019-06-18:
Change made in v01-06
The new model is as follows:

[image]

Categories names were changed as follows:

[image]

Image names were changed as follows (v.14.00)

[image]

closed 2019-09-17 by strambc