WU-BIMAC/MicroMetaApp-React #33: Re-sizing issue

strambc, 2019-06-11:
When resizing a component on the GUI, the whole window starts jittering.

strambc, 2019-06-11:
This might have been a silent crash...
When the application was relaunched it did not reproduce.

Akramhar, 2019-06-12:
You can see if there is a "crash" or problem by keeping the dev-tool opened (view -> toggle developer tools )
If red writings show up there then u can even copy and paste it in a text file and attach it here :)

Akramhar, 2019-06-17:
From other resizing issue : Huge issue with resizing: sometimes when trying to increase size it collapses down and disappears.

Akramhar, 2019-06-17:
From other resizing issue : At the moment the position of the components is lost and when the microscope is opened it looks horrible.

Akramhar, 2019-11-25:
this should not be relevant anymore

closed 2019-11-25 by Akramhar