WU-BIMAC/MicroMetaApp-React #34: Issues adding Transmittance and Reflectance range elements

strambc, 2019-06-11:
Dichroic:
entered values in metadata fields
entered values in default TR element
added 2 more TR elements + 1 RR element --> all previously entered metadata values got erased!!!

Akramhar, 2019-06-12:
Known issue, forms needs due to the fact that adding sub-components forces me to recreate all forms, hence losing the not - saved data (see #7).
To solve this issue i would have to make a fake temporary save of all the data, recreate the form and repopulate. It's doable but its a lot and needs to be well done to avoiding having dirty stuff around at all time.
I personally consider it a nice to have enhancement for a future release, for the moment i would say we just have to warn the user that it will happen if they dont save first.

Akramhar, 2019-06-17:
Add a warning under the form title if edit components is enabled

closed 2019-06-17 by Akramhar