WU-BIMAC/MicroMetaApp-React #37: Issue re-loading microscope

strambc, 2019-06-12:
After a full day yesterday in which I saved repeatedly the TESM microscope at the end of the day I moved it to Microscopes (as I had done before).
This morning when I got to the point of loading from file, it did not present a menu as usual.
It presented instead the following:
[image]

When I then loaded the TESM microscope I had saved yesterday, all most recent ELEMENTS appeared as RED.
I opened one of them and all fields were filled, but it would not allow to Confirm.

What is going on?

Attached is the Microscope file in question

[2019-6-11_microscopestesm.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3282176/2019-6-11_microscopestesm.txt)

Akramhar, 2019-06-12:
I did a fast test with the NON-standalone version, the problem seems to be in the Unit Enumeration.
When reloading, even if the unit was selected it was marked as non selected. I'm guessing it may be a problem if you selected units that have symbol inside that are currently non readable.

Akramhar, 2019-06-12:
For what concern that screen, that's one of the option to load files - from scratch / from dropped file / from folder (using the standalone version) - u may have to scroll down sometimes to see the microscopes.
If u confirm you can't see it in the scroll menu then I'll investigate testing it on a standalone version.

strambc, 2019-06-13:
I can confirm that the dropped down file list had disappeared from the screen above

strambc, 2019-06-13:
In order to further investigate I did the following test:
1) deleted MicroscopyMetadataTool folder
2) deleted alpha_0.0.5 app
3) un-compressed zip from scratch
4) launched app
5) new folder is created
6) saved test microscope
7) test microscope is saved in the parent folder
8) moved to microscopes
9) when creating a new microscope the three options are visible: scratch, file, dropdown list of existing files.
 
Now when I put the saved tesm microscope  in the newly created microscopes folder, it appears to be working again.

Akramhar, 2019-06-17:
is related to this #28

Akramhar, 2019-06-17:
the part related to #28 should be solved in 0.0.6

Akramhar, 2019-11-25:
@strambc is this still relevant?

strambc, 2019-11-27:
Appears to be solved

closed 2019-11-27 by strambc