WU-BIMAC/MicroMetaApp-React #38: Confirm: does not work even when all fields are filled in

strambc, 2019-06-13:
Other problem: does not let Confirm even when all fields are filled in.
This happened when going in and editing an existing Dichroic

strambc, 2019-06-13:
might be related with #36

strambc, 2019-06-17:
When it happens again send saved microscope.

strambc, 2019-09-17:
This problem happened again with CCD Camera in TESM Microscope.

USE CASE:
Use Electron version.
Open from repository --> CCD Camera appears red-boxed

[tesm_mma_el_91719.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3623272/tesm_mma_el_91719.txt)

When open and hit confirm the GUI jumps to 

[image]

[image]

strambc, 2019-09-17:
**MICROSCOPE FILE**
Before change
[tesm_mma_el_91719.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3623272/tesm_mma_el_91719.txt)

After change
[tesm_mma_el_91719.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3623414/tesm_mma_el_91719.txt)

strambc, 2019-09-17:
**SECOND USE CASE:**
Use the OMERO version.
Open the tesm_mma_el_91719.json (after change; see above) from file  --> CCD Camera appears red-boxed

[image]

When open and hit confirm the GUI jumps to CCD tab:

[image]

Going back to look the Camera tab the PixelWidthUnit and PixelHeightUnit fields now are filled in with "Ym" and show the "should be equal to one of the allowed values"  error.

[image]

When the field values are corrected to "µm":
[image]

and the "Confirm" button is clicked, the GUI advances to the RelayLens tab and the window does not close:

[image]

When you hit Cancel the red-box and the error above are still there.

[image]

strambc, 2019-09-17:
Might be connected with #28, with #37 and with #41

strambc, 2019-09-17:
***MICROSCOPE FILE***
After loading on OMERO and saving

[TESM_MMA_el_91719.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3623551/TESM_MMA_el_91719.txt)

Akramhar, 2019-11-25:
@strambc  is this still relevant?

Akramhar, 2019-11-27:
i think most of this should be fixed in 0.18, should be noted that some stuff have been reworked and the components that don't currently validate should be removed and re created due to some changes in the code

strambc, 2019-11-27:
Seems solved

closed 2019-11-27 by strambc