WU-BIMAC/MicroMetaApp-React #45: Error with saving: Dichroic does not save content of fields

strambc, 2019-09-11:
Does not save CutIn, CutOut, and Transmittance or TR elements in Dichroic.
When hitting Confirm, it allows closing, the red box goes away, but the content of fields is not retained.

Opened in TIer5
Validate in Tier5

[tesm_mma_ome_9719.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3603071/tesm_mma_ome_9719.txt)

[image]
[image]

strambc, 2019-09-11:
Software version: alpha_0.0.15b

strambc, 2019-09-11:
Removed problem Dichroic
Created a new one.
Same issue presented again.

[image]

strambc, 2019-09-11:
Attempted to fix by removing RR elements in the same Dichroic component.
Did not work.

REMOVED Dichroic and did not create any RR components (just 3 TR components) and:

strambc, 2019-09-12:
Created new Dichroic element (3 TR and 3 RR components) and the same problem presented.
Realized that what happens is that the CutIn and CutOut values in the corresponding RR components get copied over to the TR (i.e. RR1 --> TR1 etc.).
However the same does not happen with Reflectance vs. Transmittance. The value in Transmittance gets erased and not replaced with the value in Reflectance.

Attempted to remove RR elements and the same problem persisted.
Attempted to create a new Dichroic without any RR components (just 3 TR) and the issue went away.

Akramhar, 2019-11-26:
should be fixed in 0.18

closed 2019-11-27 by strambc