WU-BIMAC/MicroMetaApp-React #47: Removing Dichroic does not invalidate FilterSet

strambc, 2019-09-11:
When removed Dichroic 1 the Filteset that referred to Dichroic 1 did not get invalidated (the red box did not appear).
When editing the component, now it does not allow to Confirm.

SHOULD GIVE WARNING THAT CORRESPONDING ELEMENT HAS BEEN REMOVED

strambc, 2019-09-11:
I do not think this is a bug.
However the feature should be added.

Akramhar, 2019-11-27:
the microscope now should get updated correctly when a link is broken.
there is no warning on window if the field was not required to begin with.
if the field was require it should fail to validate and consequently the red border should appear.
i'm not totally sure about this part due to the fact we don't have required link on the model atm.

strambc, 2019-11-27:
[image]

strambc, 2019-11-28:
Let's discuss after next week

closed 2024-03-26 by strambc