WU-BIMAC/MicroMetaApp-React #51: Issue with PiezoElecticObjective focusing device

strambc, 2019-09-17:
USE CASE

Created TESM microscope (tesm_mma_el_91719) using Electron stand-alone version.
PiezoElecticObjective --> "ZTravelRangeUnit" was assigned "µm".
[image]

TESM was saved as: 

--> [tesm_mma_el_91719.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3623125/tesm_mma_el_91719.txt)

When this was opened in OMERO the "PiezoElectricObjective " appeared as un-validated (red boxed), when clicking Confirm the GUI advances to the "Sample Positioning" tab and shows that in the "ZTravelRangeUnit" field was assigned "Ym"
[image]
[image]

strambc, 2019-09-17:
When the "ZTravelRangeUnit" field was corrected to "µm", the microscope was saved and it was re-opened then the problem goes away

[image]

strambc, 2019-09-17:
When the PiezoElecticObjective --> "ZTravelRangeUnit" field is assigned "mm", the problem does not appear:

[image]

Akramhar, 2019-11-26:
this seems to be fixed in 0.18

strambc, 2019-11-28:
Need to test when v18 is in OMERO

closed 2020-08-19 by strambc