WU-BIMAC/MicroMetaApp-React #56: v0016 Component jump after dragging

strambc, 2019-10-30:
Electron version v0016
Microscope tesm_mma_el_91719.json
[tesm_mma_el_91719.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3786927/tesm_mma_el_91719.txt)

Opened the microscope from repository
Scrolled all the way right and attempted to reposition one of the "LightSources"

[image]
[image]

Akramhar, 2019-10-30:
Seems the mac version didn't get the upgrade
Gonna release another one right now

strambc, 2019-10-31:
Ok this is fixed

closed 2019-10-31 by strambc