WU-BIMAC/MicroMetaApp-React #57: Resizing does not get saved

strambc, 2019-10-31:
v0016-1
Electron stand-alone for Mac
tesm_mma_el_103019.json
[2019-06-17_Strambio_NewEgg order.pdf](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3791955/2019-06-17_Strambio_NewEgg.order.pdf)

When items are resized, microscope saved, back and re-open
the items do not retain the new size but go back to the original size

strambc, 2019-10-31:
Update: it appears that in both Electron and OMERO version the saving of re-sizing issue is limited to making an icon smaller.
It will save making an item bigger but not smaller.
In OMERO when the microscope is initially loaded the correct, smaller size is displayed for a fraction of a second and then it resizes back to default.

[tesm_mma_el_103019.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3795438/tesm_mma_el_103019.txt)

Akramhar, 2019-11-26:
this should be fixed in 0.18, correct?

closed 2019-11-28 by strambc