WU-BIMAC/MicroMetaApp-React #58: Sporadic component jumping issue

strambc, 2019-10-31:
v0016-1
Electron stand-alone for Mac

Ex-Filter - Empty

strambc, 2019-10-31:
Happened again with Cube1Dich
Jumped down and left
Same happened with one of the Filter wheel Em-Filters
Then it got restored

strambc, 2019-10-31:
Happened again with FilterSet Cube 6
and then with TESM Cube6Em

[image]
[image]

strambc, 2019-10-31:
[2019-06-17_Strambio_NewEgg order.pdf](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3792063/2019-06-17_Strambio_NewEgg.order.pdf)

strambc, 2019-10-31:
Really difficult to decipher
One common theme is that when it starts jumping then it jumps a couple more times until it is moved to a "safe" position and then it stops jumping

strambc, 2019-10-31:
It happened 7 times in one session

strambc, 2019-10-31:
Microscope
microscope_template_2.json

Moved Relay Lens and it jumped all the way down as if the canvas was twice as high and microscope was in the top left

Akramhar, 2019-11-26:
this needs to be verified in 0.18 to see if it's still relevant

strambc, 2019-11-28:
Did not observe yet but will leave it open until testing more extensively

closed 2020-08-19 by strambc