WU-BIMAC/MicroMetaApp-React #60: Validation Issue #3

strambc, 2019-10-31:
v0016-1
[tesm_mma_el_103019.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/3795740/tesm_mma_el_103019.txt)

Microscope was saved yesterday and reopened a number of times on Electron.
Validation issue with Camera (which had been ongoing) appeared to be fixed.

When same file opened on OMERO the Camera + PiezoElectric Focus showed up as non-validated (red boxed)

strambc, 2019-10-31:
[image]
[image]
[image]
[image]
[image]

strambc, 2019-11-08:
[image]

Akramhar, 2019-11-27:
should be fixed in 0.18

strambc, 2019-11-28:
Have to wait until v18 is on OMERO to test

closed 2020-08-19 by strambc