WU-BIMAC/MicroMetaApp-React #61: TESM_MMA_el_110619 issues

strambc, 2019-11-27:
The TESM_MMA_el_110619 microscope that was previously saved appears to have issues with the size of the BOUNDING BOXES which do not resize according to the new sizes but retain the old size (see example)
[image]

WOULD IT BE POSSIBLE TO FIX? 
If not I will have to rebuild the microscope from scratch (which might be a problem as I would like to use the screenshot for the poster)

strambc, 2019-11-27:
The issue is not present with newly added elements:

[image]

Akramhar, 2019-11-27:
if you resize the element to the max size it should get fixed automatically

closed 2019-11-28 by strambc