WU-BIMAC/MicroMetaApp-React #65: Resizing bug: when trying to resize the new size gets stuck

strambc, 2019-11-29:
[image]
[image]
[image]

strambc, 2019-11-29:
**Examples:**

[image]

[image]

[image]

Akramhar, 2019-11-30:
there is a chance this may be fixed in 0.0.19, needs testing

closed 2019-12-02 by strambc