WU-BIMAC/MicroMetaApp-React #66: Objective: new model was not incorporated

strambc, 2019-11-30:
was v01-07 of the model incorporated?
There were changes in Objective that were not incorporated.

strambc, 2019-12-02:
Appears to be incorporated now

strambc, 2019-12-02:
However, I noticed the description of CorrectionCollar is not all there. 

**CorrectionCollar** 
The current description is:
"Records the type of Correction Collar this objective is fitted with."

The updated description should be:
"When using high numerical aperture dry objectives (numerical aperture of 0.8 or greater), cover glass thickness variations of only a few micrometers, immersion medium changes, and temperature changes result in dramatic image degradation due to aberration. To compensate for these errors, the more highly corrected objectives are equipped with a correction collar to allow adjustment of the central lens group position to coincide with fluctuations in cover glass thickness, changes in immersion medium and in temperature. This field records whether or not the objective was fitted with a Correction Collar."

Was this on purpose?

Akramhar, 2019-12-02:
In the xml file this is what i get: <xsd:documentation>Description=Records the type of Correction Collar this objective is fitted with.</xsd:documentation>

Akramhar, 2019-12-02:
I made a hardcode gen to avoid having to regenerate the model for this round. Is the updated description in 1.07 or 1.08?

strambc, 2019-12-02:
It is v07 and I checked. The description was not there.
It is there now.
C

Akramhar, 2019-12-03:
should be fixed in 0.21

closed 2020-06-22 by strambc