WU-BIMAC/MicroMetaApp-React #67: All elements have Add/Remove Band Pass

strambc, 2019-11-30:
Was this done on purpose?

Akramhar, 2019-11-30:
I tested it and the button to edit band-pass seems to be disabled for the elements that are not supposed to have band-pass. Are you experiencing something different?

strambc, 2019-11-30:
Yes it is dibbled but I was wondering whether this was on purpose. I do not think it was there before…
Right?

Akramhar, 2019-11-30:
Looking at the code, it should have, I didn't change how the buttons are created/positioned. I changed only how to enable/disable that specific button.

strambc, 2019-12-02:
OK so now all elements have it and it is disabled if not relevant. Correct?
Is this how we want to have it?

Akramhar, 2019-12-02:
we can try to remove it when not relevant in the next development round

strambc, 2020-06-22:
Would remove it if possible

closed 2021-01-07 by strambc