WU-BIMAC/MicroMetaApp-React #69: Issue with different scalingFactors

Akramhar, 2019-12-03:
If a mic is created on a system with scalingFactor 0.5 and its then opened on a system with scalingFactor 1 the images are not rescaled correctly.
Same is true for a microscope without scalingFactor reopened on a system where a scalingFactor different from 1 is applied.

Possible solution: check for a scalingFactor on the microscope, if not present assume 1. compare to the current system scalingFactor. if different, remove all width/height previously set.
This would reset the image size and force a recompute with the newly applied scalingFactor.
Could be the fastest route to solve the issue, but at the same time this would lose all the user made resizing due to the fact we currently don't save single component user set scale.
Need to evaluate pros and cons better.

Akramhar, 2019-12-06:
This may be solved in 0.22

strambc, 2021-01-06:
Related to #104

closed 2021-07-14 by strambc