WU-BIMAC/MicroMetaApp-React #70: SVG & PNG Image Full Path Not Calculated Properly

utku-ozturk, 2019-12-09:
SVG & PNG image full path is miscalculated for the following case:

`ImagesPathPNG` and `imagesPathSVG` are provided explicitly via props instead of default './assets/png/' (e.g. https://raw.githubusercontent.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/master/public/assets/svg/) And 4DNMicroscopyMetadataToolReact component is hosted a site that uses https. (e.g. https://www.example.com)

Then, the full url path has a missing "/": `https:/` instead of `https://` (e.g. https:/raw.githubusercontent.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/master/public/assets/svg/Transmitted_LightSource_GenericExcitationSource.svg?sanitize=true)

On the other hand, if the site uses `http` then image full path is calculated properly.

The possible solution would be 

1. url.resolve instead of path.join while combining full url from base path and file name. https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/blob/087c6bc67a057529b1c8cb9a045e5b41b4e43c13/src/components/toolbar.js#L83
2. simple string concatenation 

`url.resolve` and `path.join` url combine diff. sandbox here, pls check : https://codesandbox.io/s/utku-ozturk-wyshr

Akramhar, 2019-12-11:
this should be solved in version 0.22
one note is that you need to setup the path with a / at the end otherwise url.resolve doesn't work properly (see [here](https://github.com/nodejs/node/issues/8921))

utku-ozturk, 2019-12-12:
thanks @Akramhar, I'll test shortly and inform you.

utku-ozturk, 2020-02-09:
@Akramhar image.path left on this line and should be replaced w/ url. resolve: https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/blob/406e7cd8840a914cfd7e31bee91a4cc79b820910/src/components/toolbar.js#L83

Akramhar, 2020-02-11:
#71 should solve this, please let me know so I can close this issue :)

utku-ozturk, 2020-02-11:
Will update the package in FF and if all set after deployment, then we would close the issue. I'll inform you @Akramhar.

Akramhar, 2020-06-13:
@utku-ozturk was this solved? Can we close it?

closed 2020-08-26 by Akramhar