WU-BIMAC/MicroMetaApp-React #75: Button rename

strambc, 2020-06-02:
Create Microscope >> Manage Instrument Hardware or Manage Hardware
Use Microscope >> Manage Image Acquisition Settings or Manage Acquisition Settings or Manage Settings

Choose one of the options depending on whether or not the text fit in the button without totally destroying the estetic of the GUI.

strambc, 2020-06-02:
Validate @ tier: 1 >> Validate @ Tier 1
Save microscope >> Save Microscope file

Akramhar, 2020-06-13:
I wrote on my notes "Manage Instrument" and "Manage Settings", i added that in version 0.23, let me know if that works.

strambc, 2020-06-13:
Ok works

Caterina Strambio De Castillia, Ph.D. | CZI Imaging Scientist | Assistant Professor, Program in Molecular Medicine | University of Massachusetts Medical School |
——————————————
Please excuse any spelling errors or short replies, this message was sent to you from my iPhone

Akramhar, 2020-06-17:
everything in this issue should be included in 0.23

closed 2020-06-22 by strambc