WU-BIMAC/MicroMetaApp-React #76: Mouse over text to add

strambc, 2020-06-02:
### **Window 1**

**Browse** button >> "Click this button to navigate your file system and select the home folder where your Microscope files will be saved. This folder will constitute your local Repository."

**Confirm** button >> "Click Confirm to continue."

###  **Window 2**
**Tier selector** >> "Choose the Tier level you want to utilize to document Instrument Hardware Specifications and the Image Acquisition Settings. Please refer to the following link for more details about Tier definition: https://arxiv.org/ftp/arxiv/papers/1910/1910.11370.pdf"
**Manage Instrument Hardware** button >> "Click this button to create, load from a template, or edit a Microscope file containing the configuration and specifications of the hardware components of the Instrument you want to document".
**Manage Acquisition Settings" button >> "Click on this button to select an existing Microscope file and enter, load from file, or edit the acquisition settings associated with a specific Image dataset you want to document".

### **Window 3 (Manage Instrument Hardware)**
**Create from scratch** selector >> "Choose the modality you want to use for Instrument hardware management. 'Create from scratch' allows you to create a brand new Microscope file from scratch. 'Load from file' allows you to import into Micro-Meta App a previously available Microscope file (i.e., an example file, a template file, or a file shared by a colleague) to edit. 'Load from repository' allows you to load a previously available file from the active Micro-Meta App repository."
**Confirm** button >> "Click Confirm to continue."
**Back** button >> "Click Back to go back to the previous screen."

strambc, 2020-06-14:
"Hardware explorer"  menu title >> "Hardware component selection: Use this menu to select hardware components to include in this Microscope file. Please follow these steps: 1) Open an individual menu. 2) Drag-and-drop the desired element into the main canvas. 3) Click on the desired element to enter the information you want to document. 4) Click on Confirm to save the information you have entered.

Akramhar, 2020-06-16:
in window 3 we are missing the following tooltips:
-for the dropzone - where you click to load a file after you select 'load a file'
-for the manufactorer selector - where you click to select a manufacturer after you select 'load from repo'
-for the microscope selector - where you click to select a microscope after you select 'load from repo' and a manufacturer

strambc, 2020-06-16:
Create from file "drop zone" tool-tip >> Select an existing Microscope file you want to work on.
Create from file "drop zone" text >> Select an existing Microscope file you want to work on.
Manufacturer selector >> Select the Manufacturer of the Microscope you want to load.
Microscope selector >> Select the Microscope file you want to load.

Akramhar, 2020-06-17:
everything in this issue should be included in 0.23

strambc, 2020-06-22:
Window 1: tooltips missing
Window 2: 
Tier selector >> Tooltip is not correct (see text above).
Continue >> Title: from "Create mode Continue" to "Continue"

closed 2020-06-22 by strambc