WU-BIMAC/MicroMetaApp-React #86: Tool tips issues

strambc, 2020-06-22:
Window 2: 
Tier selector >> Tooltip is not correct (see text above).
Continue >> Title: from "Create mode Continue" to "Continue"

strambc, 2020-06-22:
Related to #76

Akramhar, 2020-07-13:
Fix the "continue".
The only difference with the "Tier selector" is regarding the link part, that I aligned with the "Validation tier".
Is that the problem you are referring too?

strambc, 2020-08-19:
ISSUE 1

Tier selector --> tooltip text has a typo.
The correct text of the first sentence should be: "Choose the Tier level you want to document Instrument Hardware Specification and Image Acquisition Setting"
The second sentence is OK as it is.

ISSUE 2
In the Tier validation selector, Tier should be spelled with a capital T

Akramhar, 2020-08-26:
should be fixed in 0.28

closed 2020-08-26 by strambc