WU-BIMAC/MicroMetaApp-React #87: v0025 >> BLANK SCREEN ISSUE

strambc, 2020-06-22:
it happened again in v0025

strambc, 2020-06-22:
Uncaught (in promise) Error: ENOENT, microscopes not found in /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar
    at createError (electron/js2c/asar.js:109)
    at Object.fs.readdir (electron/js2c/asar.js:630)
    at eval (webpack://MicroscopyMetadataToolStandAlone/./src/app.js?:45)
    at new Promise (<anonymous>)
    at readDirAsync (webpack://MicroscopyMetadataToolStandAlone/./src/app.js?:44)
    at Function.copyFiles (webpack://MicroscopyMetadataToolStandAlone/./src/app.js?:217)
    at MicroscopyMetadataToolComponent.componentDidMount (webpack://MicroscopyMetadataToolStandAlone/./src/app.js?:305)
    at commitLifeCycles (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19814)
    at commitLayoutEffects (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:22803)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
webpack://MicroscopyMetadataToolStandAlone/./node_modules/url/url.js?:112 Uncaught TypeError: Parameter 'url' must be a string, not undefined
    at Url.parse (webpack://MicroscopyMetadataToolStandAlone/./node_modules/url/url.js?:112)
    at urlParse (webpack://MicroscopyMetadataToolStandAlone/./node_modules/url/url.js?:106)
    at Url.resolve (webpack://MicroscopyMetadataToolStandAlone/./node_modules/url/url.js?:443)
    at Object.urlResolve [as resolve] (webpack://MicroscopyMetadataToolStandAlone/./node_modules/url/url.js?:439)
    at MicroscopyMetadataTool.render (webpack://MicroscopyMetadataToolStandAlone/./node_modules/4dn-microscopy-metadata-tool/es/app.js?:1195)
    at finishClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17160)
    at updateClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17110)
    at beginWork (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:18620)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527 The above error occurred in the <MicroscopyMetadataTool> component:
    in MicroscopyMetadataTool (created by MicroscopyMetadataToolComponent)
    in MicroscopyMetadataToolComponent

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
logCapturedError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527
[... cut ...] .25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:12639
Component.setState @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/alpha_0.0.25/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react/cjs/react.development.js:471
onWindowResize @ app.js:262

strambc, 2020-06-23:
Both "dimensions" and "schemas" became empty

Akramhar, 2020-08-02:
This may be fixed in 0.0.26

closed 2020-08-19 by strambc