WU-BIMAC/MicroMetaApp-React #90: Em and Ex Filter

strambc, 2020-06-23:
Em and Ex Filter icons have too much border around them

[image]

Is this something that has to be fixed by Alex or Shannon?

strambc, 2020-06-23:
**Intensity Calibration Tool**
[image]

strambc, 2020-06-23:
**AutoFocus**
[image]

strambc, 2020-06-23:
**FocusingObjectiveTurret**
[image]

strambc, 2020-06-23:
**Microscopy Table**
[image]

strambc, 2020-06-23:
**Optovar**
[image]

strambc, 2020-06-23:
**Stage and Sample Holder**
[image]

strambc, 2020-06-23:
**Additional Optics + Mirror**
[image]

Akramhar, 2020-07-13:
IntensityCalibrationTool, LightSourceCoupling and ObjectiveTurretFocus had issues with the pre-set dimensions and have been fixed in 0.26.
For what concern the other element I have been unable to reproduce the error of having elements with the "squared" bounding box. Could you please tell me more about how you obtained this issue?
For what concern the additional optics / mirror I'm unable to see what the problem related bounding box would be.

strambc, 2020-08-19:
-- PROBLEM SOLVED WITH ALL STAGE ISSUE AND WITH CALIBRATION AND AUTOFOCUS
-- PROBLEM PARTIALLY PERSISTS WITH ALL OTHER ELEMENTS

PARTICULARLY NOTICEABLE WITH MICROSCOPE TABLE

Akramhar, 2020-08-26:
may be solved in 0.28

strambc, 2020-08-26:
-- PROBLEM SOLVED WITH TABLE, OPTOVAR, FOCUSING TURRET
-- PROBLEM PERSISTS WITH EX-FILTER, EM-FILTER, COUPLING LENS, RELAY LENS

Akramhar, 2020-08-26:
should be fixed in 0.28.2-a1

strambc, 2020-08-26:
-- PROBLEM FIXED WITH EM-FILTER
-- PROBLEM WITH EX-FILTER, COUPLING AND RELAY LENS PERSISTS

strambc, 2020-08-26:
ZOOMED IN

[image]

ZOOMED OUT

[image]

strambc, 2020-08-26:
ALSO WEIRD THAT COUPLING AND RELAY ICONS HAVE DIFFERENT SIZES!

Akramhar, 2020-08-27:
thats how they are done by shannon
should be fixed in 0.28.3-a1

closed 2020-12-28 by strambc