WU-BIMAC/MicroMetaApp-React #93: Resizing issues

strambc, 2020-06-23:
**CORRECT RESIZING**
THE PROPORTION OF THE BOUNDING BOX WITH RESPECT TO THE ICON DOES NOT CHANGE
[image]

strambc, 2020-06-23:
**WRONG RESIZING**
THE PROPORTION OF THE BOUNDING BOX WITH RESPECT TO THE ICON CHANGES WITH SIZE
[image]

strambc, 2020-06-23:
THIS SAME ISSUE APPLIES TO:

- LightSourceCoupling _ Light Guide
- Em-Filter
- Objective Turret Focus
- Stage and Sample Holder

Akramhar, 2020-07-13:
This should be fixed for LightSourceCoupling, TurretFocus, IntensityCalibrationTool.
For what concern Excitation and Emission Filter I'm unable to reproduce similar behavior, the one I'm aware so far is currently related to the snap in place.
Stage and SampleHolder I'm unable to reproduce and I'm not aware of any related issue.
Could you please describe the issue more in detail?

closed 2020-08-19 by strambc