WU-BIMAC/MicroMetaApp-React #94: Icons disappears upon resizing

strambc, 2020-06-23:
This happened upon re-opening an existing microscope

25_microscope_2.txt

[25_microscope_2.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/4821026/25_microscope_2.txt)

strambc, 2020-06-23:
[image]
[image]
[image]
[image]

strambc, 2020-06-23:
At the same time PRISM resized correctly as below
[image]

strambc, 2020-06-23:
25_Microscope_2 was reloaded and the issue was seen again.

strambc, 2020-06-23:
When a new empty microscope is created using Create from scratch the behavior does not happen.
When the new microscope was saved (25_Microscope_3) was saved and reopened the same behavior happened again.

[25_microscope_3.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/4821078/25_microscope_3.txt)

Akramhar, 2020-07-13:
I tried to load the microscope you posted.
It seems to be loading correctly:
[image]
Could you please tell me more about the issue?

closed 2020-08-19 by strambc