WU-BIMAC/MicroMetaApp-React #97: Validate microscope only when a minimal set of elements are present

strambc, 2020-08-19:
At the moment one can validate an empty microscope.
This has to be changed.
Let's discuss

strambc, 2021-07-14:
Add "MinimumComponentsList" field element in "tierList.json" file

strambc, 2021-07-14:
Will be done

strambc, 2025-03-18:
Example Minimum Component List:

MinimumComponentsList = "LightSource.json", "Objective.json", "Detector.json", ["Filter.json", "Dichroic.json"]

Akramhar, 2025-03-18:
Add min components to the tierlist.json.
Load the min components when tierlist gets loaded.
Add list in canvas under the canvas labels in red or black on red background with the list of the missing components to reach minimum . Delete missing components when said components type gets added (keep updated on each drop, probably only if minimum not reached yet ?)

strambc, 2025-03-18:
Proposed solution

[image]

strambc, 2025-09-17:
Added "MinimumComponentsList" field element in "tierList.json" file

https://github.com/WU-BIMAC/4DNMetadataSchemaXSD2JSONConverter/blob/master/versions/2-01-1/tiers/TierList.json