WU-BIMAC/MicroMetaApp-React #99: Issues from Jim Chambers

strambc, 2020-09-29:
- Seems I cannot point it to a custom folder on startup…it defaults to the one it wants even if I point it to my test folder on my Win10 desktop.

- Laser light source Which nominal power output to report if it is a multiple (combined) laser cabinet?  Plays into my next question. Or should each laser line be it’s own “laser”?

- LightSourceCoupling
How to do it for 3 different sources?  I have an LED for epiillumination, a Nikon laser cabinet that is fiber coupled, and I also have 4 pulsed lasers fiber coupled for FLIM/FCS.
After saving work, it seems I cannot edit any components.  I need to reload the application.
Objective question
I assume Achroplan = Plan Apochromat?  I think Nikon uses slightly different nomenclature.
Is there a way to copy and past parts?  I have 4 PMTs, for instance.  And 8 lasers…all similar.  Would be nice to copy and paste and then just edit for the differences if possible?

strambc, 2020-12-28:
@Akramhar was this fixed?

Akramhar, 2021-03-16:
solved in 0.39

closed 2021-03-16 by strambc