WU-BIMAC/MicroMetaApp-React #104: Compatibility issue with different version (electron, plugin, etc)

Akramhar, 2020-10-09:
If a microscope is saved with one version and then opened with a different version, there is the chance of having misalignments and snap-in-place issues.
One solution could be saving the offset used to save the microscope in the microscope itself (and consequently in the json file created) and then use them during import/loading to compare against current offset. If the offset are different a new position for component should be computed based on the difference.
Needs to be verified and tested before implementation.

strambc, 2021-07-14:
This requires some testing to see how this impacts the capacity of users to create a file in Electron and then import it on the portal etc.

strambc, 2021-07-14:
This is related to now closed issue #69