WU-BIMAC/MicroMetaApp-React #111: Any entry that is yes/no requires you to have the checkbox marked

paulamonterollopis, 2020-11-23:
It seems in any entry that has a checkbox to check only if the entry is true (e. g tunable laser), the software treats it as not answered and required. Perhaps 2 checkboxes should be available? 

I think it may be a bug

strambc, 2020-11-23:
Dear Paula
I think this is a known issue.
Can you please try to check and un-check it and see if it now accepts it?

paulamonterollopis, 2020-11-23:
Yes, it now accepts it no problem.

Thank you.

Paula

Paula  Montero Llopis, PhD

MicRoN Director

77 Avenue Louis Pasteur

NRB1032

Office 617 432 0177

Cell 203 927 1468

micron.hms.harvard.edu

strambc, 2020-11-23:
Great so this is a known issue we will fix in an upcoming release. The way it is is really ANNOYING and it has been noticed before.

strambc, 2020-11-23:
This is related to #105

strambc, 2020-11-23:
I am going to close this issue as it a duplicate of issue #105

closed 2020-11-23 by strambc

paulamonterollopis, 2020-11-23:
Oh, I am sorry! I think this is sort of what was happening with Joel and the environmental chamber 🙂

Paula  Montero Llopis, PhD

MicRoN Director

77 Avenue Louis Pasteur

NRB1032

Office 617 432 0177

Cell 203 927 1468

micron.hms.harvard.edu

strambc, 2020-11-23:
No problem at all. Thank you for pointing it out!!!!