WU-BIMAC/MicroMetaApp-React #115: Montero#4 Version Micro Meta App v0.29.0-a1 does not appear to work with macOS Catalina

paulamonterollopis, 2020-12-15:
I tried downloading Micro Meta App v0.29.0-a1 into my Mac following the detailed Downloading instructions. When I try running opening the app after extracting the files, I get this error message:
"4dn-microscopy-metadata-tool-standalone cannot be opened because of a problem"
"Check with the developer to make sure 4dn-microscopy-metadata-tool-standalone works with this version of macOS. You may need to reinstall the application. Be sure to install any available updates for the application and macOS"

Thank you!

strambc, 2020-12-15:
Dear Paula
This is strange. 
It works on my Mac.
I have Catalina 10.15.7
What MacOS version do you have?
C

paulamonterollopis, 2020-12-15:
I may be doing something weird. It is the same version you have. Let me see if I can figure out what's going on!

Thank you!

Paula

Paula  Montero Llopis, PhD

MicRoN Director

77 Avenue Louis Pasteur

NRB1032

Office 617 432 0177

Cell 203 927 1468

micron.hms.harvard.edu

strambc, 2020-12-15:
No you are not doing something weird.
This is something we have seen before but we cannot nail down.
So the more cases we see the better
I have sent you an email with a link to an un-zipped version that works on my computer.
Let me know if that works.

paulamonterollopis, 2020-12-15:
Thank you Caterina.

I am deleting all MicroMetaApp files from my computer and start afresh with the link you sent me. I wonder if this is due to some security issue, HMS tends to be very conservative,

Thanks!

Paula

paulamonterollopis, 2020-12-15:
Sigh, it does not work. Usually when I download apps in my mac it prompts you to double click on them, but this one doesn't do it. I went to privacy settings in case the computer was blocking it, but it didn't work either. I am unsure of what else to try...

Thanks!

Paula

Paula  Montero Llopis, PhD

MicRoN Director

77 Avenue Louis Pasteur

NRB1032

Office 617 432 0177

Cell 203 927 1468

micron.hms.harvard.edu

strambc, 2020-12-15:
What is your MacOS version?
C

paulamonterollopis, 2020-12-15:
It's 10.15.7

I think it's the same as yours.

Thank you, Caterina!

P

Paula  Montero Llopis, PhD

MicRoN Director

77 Avenue Louis Pasteur

NRB1032

Office 617 432 0177

Cell 203 927 1468

micron.hms.harvard.edu

strambc, 2020-12-15:
I am sorry about this.
I have tested what happens on my hand and un-zipped again the Mac archive on my computer.
Mysteriously, it does not work and gives me the same issue you are getting.
So I am really confused.
Could you please do me a favor?
Could you try re-starting the computer and see if anything changes?
Do not need to do it now, you can wait to re-start some other time and let me know later.
Cheers

paulamonterollopis, 2020-12-15:
I restarted the computer and erased all previous downloads before trying the last version you sent,  just in case it was my computer but I didn't see a change.

What I did notice is that when I download from Dropbox it downloads at zip file. I wonder if it is a dropbox problem.

Thank you

Paula  Montero Llopis, PhD

MicRoN Director

77 Avenue Louis Pasteur

NRB1032

Office 617 432 0177

Cell 203 927 1468

micron.hms.harvard.edu

strambc, 2020-12-15:
OK I found a possible fix.
This is what I did:

Right-clicked (or Control-click) on the zip file.
Chose Open with > The Unarchiver.app
Installed all Unarchiver.app updates that were suggested
Unarchived the zip
Right-clicked (or Control-click) on the Micro-Meta App file
Click on “Open”

This allowed me to launch the freshly unzipped file again.

Let me know if this works for you.

paulamonterollopis, 2020-12-15:
AHHH, it does not work for me, I don't understand!

I just downloaded the app (I didn't have it). Let me download again

P

Paula  Montero Llopis, PhD

MicRoN Director

77 Avenue Louis Pasteur

NRB1032

Office 617 432 0177

Cell 203 927 1468

micron.hms.harvard.edu

strambc, 2020-12-16:
This is not cool.
I am sorry.
Could you send me the exact error message?

paulamonterollopis, 2020-12-16:
I am so sorry, Caterina. This is the message I got:

"4dn-microscopy-metadata-tool-standalone cannot be opened because of a problem"
"Check with the developer to make sure 4dn-microscopy-metadata-tool-standalone works with this version of macOS. You may need to reinstall the application. Be sure to install any available updates for the application and macOS"

But finally it’s working in my laptop and I was much faster at creating the configuration, after having used it before.

Thank you!

Paula

closed 2021-01-07 by strambc