WU-BIMAC/MicroMetaApp-React #117: Change Validation Tier

strambc, 2020-12-16:
Opened from scratch and changed validation tier from 3 to 1.
The application went blank.
See attached console error messages

strambc, 2020-12-16:
[
[image]
[image]
[image]
[image]

](url)

Akramhar, 2020-12-18:
this should be solved in 0.31

strambc, 2020-12-28:
Fixed

closed 2020-12-28 by strambc