WU-BIMAC/MicroMetaApp-React #119: Fatal error: Does not load from file

strambc, 2020-12-16:
Saved Microscope at Tier 1
Attempted to Load from Repository at same Tier
See attached errors

strambc, 2020-12-16:
[
[image]
[image]

](url)

strambc, 2020-12-16:
**Errors from Load from File**

[
[image]
[image]

](url)

Akramhar, 2020-12-19:
this should be fixed in 0.31.
in addition microscope from older version should be somewhat adjusted at least for what concern the MicroscopeStand and the version getting updated.
There are still issues in following cases:
1) a schema name was changed : during the load phase components whose model cannot be found are discarded
2) fields are renamed / removed : the value may persist (needs to be verified) even though they are not accessible in the editor form

closed 2020-12-28 by strambc