WU-BIMAC/MicroMetaApp-React #144: OpticalAssembly: cannot add any of the Referenced items

strambc, 2020-12-17:
Condenser:

- cannot reference existing multiple CondenserLens
- cannot reference existing multiple OpticalAperture

strambc, 2020-12-17:
Same happens with MicroscopeTube and BeamExpander and BeamCollimator

Akramhar, 2020-12-24:
In theOpticalAssembly:
The problem with Aperture is that they are called Aperture in the model but they are trying to get referenced as OpticalAperture.
The problem MirroringDevice is that they are called MirroringDevice in the model but they are trying to get referenced as MirrorringDevice :(

Akramhar, 2020-12-24:
Once they are aligned, the new system in place seems to be working properly so we should be able to reference by category name and not just by element model name

strambc, 2020-12-24:
Not sure I understand. Should I change the model? I thought we were using the category... I am confused

Caterina Strambio De Castillia, Ph.D. | CZI Imaging Scientist | Assistant Professor, Program in Molecular Medicine | University of Massachusetts Medical School |
——————————————
Please excuse any spelling errors or short replies, this message was sent to you from my iPhone

strambc, 2020-12-24:
Committed change to model

Caterina Strambio De Castillia, Ph.D. | CZI Imaging Scientist | Assistant Professor, Program in Molecular Medicine | University of Massachusetts Medical School |
——————————————
Please excuse any spelling errors or short replies, this message was sent to you from my iPhone

On Dec 24, 2020, at 06:52, Strambio, Caterina <Caterina.Strambio@umassmed.edu> wrote:

 Not sure I understand. Should I change the model? I thought we were using the category... I am confused

Caterina Strambio De Castillia, Ph.D. | CZI Imaging Scientist | Assistant Professor, Program in Molecular Medicine | University of Massachusetts Medical School |
——————————————
Please excuse any spelling errors or short replies, this message was sent to you from my iPhone

strambc, 2020-12-28:
THIS IS FIXED

closed 2020-12-28 by strambc