WU-BIMAC/MicroMetaApp-React #150: EDIT MICROSCOPE AND CONFIRM CRASH

strambc, 2020-12-28:
Νew microscope 
Edit microscope
Confirm
Goes blank
Similar to issue #149

strambc, 2020-12-28:
Error
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:88 Warning: React does not recognize the `pId` prop on a DOM element. If you intentionally want it to appear in the DOM as a custom attribute, spell it as lowercase `pid` instead. If you accidentally passed it from a parent component, remove it from the DOM element.
    in path (created by AiOutlinePlus)
    in svg (created by Context.Consumer)
    in IconBase (created by AiOutlinePlus)
    in AiOutlinePlus (created by AddButton)
    in button (created by Button)
    in Button (created by AddButton)
    in AddButton (created by DefaultNormalArrayFieldTemplate)
    in div (created by Col)
    in Col (created by DefaultNormalArrayFieldTemplate)
    in div (created by Row)
    in Row (created by DefaultNormalArrayFieldTemplate)
    in div (created by Container)
    in Container (created by DefaultNormalArrayFieldTemplate)
    in div (created by Container)
    in Container (created by DefaultNormalArrayFieldTemplate)
    in div (created by Col)
    in Col (created by DefaultNormalArrayFieldTemplate)
    in div (created by Row)
    in Row (created by DefaultNormalArrayFieldTemplate)
    in div (created by DefaultNormalArrayFieldTemplate)
    in DefaultNormalArrayFieldTemplate (created by ArrayFieldTemplate)
    in ArrayFieldTemplate (created by ArrayField)
    in ArrayField (created by SchemaField)
    in div (created by FormGroup)
    in FormGroup (created by FieldTemplate)
    in FieldTemplate (created by SchemaField)
    in SchemaField (created by ObjectField)
    in div (created by Col)
    in Col (created by ObjectFieldTemplate)
    in div (created by Row)
    in Row (created by ObjectFieldTemplate)
    in div (created by Container)
    in Container (created by ObjectFieldTemplate)
    in ObjectFieldTemplate (created by ObjectField)
    in ObjectField (created by SchemaField)
    in div (created by FormGroup)
    in FormGroup (created by FieldTemplate)
    in FieldTemplate (created by SchemaField)
    in SchemaField (created by Form)
    in form (created by Form)
    in Form (created by ForwardRef)
    in ForwardRef
    in div (created by TabPane)
    in TabPane (created by MultiTabFormWithHeaderV3)
    in div (created by TabPanelList)
    in div (created by TabPanelList)
    in TabPanelList (created by ForwardRef(Tabs))
    in div (created by ForwardRef(Tabs))
    in ForwardRef(Tabs) (created by MultiTabFormWithHeaderV3)
    in div (created by TabPane)
    in TabPane (created by MultiTabFormWithHeaderV3)
    in div (created by TabPanelList)
    in div (created by TabPanelList)
    in TabPanelList (created by ForwardRef(Tabs))
    in div (created by ForwardRef(Tabs))
    in ForwardRef(Tabs) (created by MultiTabFormWithHeaderV3)
    in div (created by MultiTabFormWithHeaderV3)
    in div (created by ModalWindow)
    in div (created by ModalWindow)
    in ModalWindow (created by MultiTabFormWithHeaderV3)
    in MultiTabFormWithHeaderV3 (created by CanvasElement)
    in CanvasElement (created by Canvas)
    in div (created by Canvas)
    in div (created by Canvas)
    in span
    in div
    in t (created by Canvas)
    in div (created by Canvas)
[... 39 lines cut ...]
    in div (created by MultiTabFormWithHeaderV3)
    in div (created by ModalWindow)
    in div (created by ModalWindow)
    in ModalWindow (created by MultiTabFormWithHeaderV3)
    in MultiTabFormWithHeaderV3 (created by Footer)
    in Footer (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataToolContainer)
    in MicroscopyMetadataToolContainer (created by MicroscopyMetadataTool)
    in MicroscopyMetadataTool (created by MicroscopyMetadataToolComponent)
    in MicroscopyMetadataToolComponent

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
logCapturedError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:11102 Uncaught TypeError: Cannot convert undefined or null to object
    at Function.assign (<anonymous>)
    at MicroscopyMetadataTool.onMicroscopeDataSave (app.js:1092)
    at Footer.onFormConfirm (footer.js:82)
    at MultiTabFormWithHeaderV3.processData (multiTabFormWithHeaderV3.js:468)
    at MultiTabFormWithHeaderV3.onSubmit (multiTabFormWithHeaderV3.js:351)
    at Form.eval (Form.js:259)
    at callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alp [... cut ...] scopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19858)
    at commitLayoutEffects (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:22803)

Akramhar, 2021-01-04:
should be solved in 0.32

closed 2021-01-05 by strambc