WU-BIMAC/MicroMetaApp-React #151: Cannot save new Microscope

strambc, 2020-12-28:
**Error**

/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:88 Warning: React does not recognize the `pId` prop on a DOM element. If you intentionally want it to appear in the DOM as a custom attribute, spell it as lowercase `pid` instead. If you accidentally passed it from a parent component, remove it from the DOM element.
    in path (created by AiOutlinePlus)
    in svg (created by Context.Consumer)
    in IconBase (created by AiOutlinePlus)
    in AiOutlinePlus (created by AddButton)
    in button (created by Button)
    in Button (created by AddButton)
    in AddButton (created by DefaultNormalArrayFieldTemplate)
    in div (created by Col)
    in Col (created by DefaultNormalArrayFieldTemplate)
    in div (created by Row)
    in Row (created by DefaultNormalArrayFieldTemplate)
    in div (created by Container)
    in Container (created by DefaultNormalArrayFieldTemplate)
    in div (created by Container)
    in Container (created by DefaultNormalArrayFieldTemplate)
    in div (created by Col)
    in Col (created by DefaultNormalArrayFieldTemplate)
    in div (created by Row)
    in Row (created by DefaultNormalArrayFieldTemplate)
    in div (created by DefaultNormalArrayFieldTemplate)
    in DefaultNormalArrayFieldTemplate (created by ArrayFieldTemplate)
    in ArrayFieldTemplate (created by ArrayField)
    in ArrayField (created by SchemaField)
    in div (created by FormGroup)
    in FormGroup (created by FieldTemplate)
    in FieldTemplate (created by SchemaField)
    in SchemaField (created by ObjectField)
    in div (created by Col)
    in Col (created by ObjectFieldTemplate)
    in div (created by Row)
    in Row (created by ObjectFieldTemplate)
    in div (created by Container)
    in Container (created by ObjectFieldTemplate)
    in ObjectFieldTemplate (created by ObjectField)
    in ObjectField (created by SchemaField)
    in div (created by FormGroup)
    in FormGroup (created by FieldTemplate)
    in FieldTemplate (created by SchemaField)
    in SchemaField (created by Form)
    in form (created by Form)
    in Form (created by ForwardRef)
    in ForwardRef
    in div (created by TabPane)
    in TabPane (created by MultiTabFormWithHeaderV3)
    in div (created by TabPanelList)
    in div (created by TabPanelList)
    in TabPanelList (created by ForwardRef(Tabs))
    in div (created by ForwardRef(Tabs))
    in ForwardRef(Tabs) (created by MultiTabFormWithHeaderV3)
    in div (created by TabPane)
    in TabPane (created by MultiTabFormWithHeaderV3)
    in div (created by TabPanelList)
    in div (created by TabPanelList)
    in TabPanelList (created by ForwardRef(Tabs))
    in div (created by ForwardRef(Tabs))
    in ForwardRef(Tabs) (created by MultiTabFormWithHeaderV3)
    in div (created by MultiTabFormWithHeaderV3)
    in div (created by ModalWindow)
    in div (created by ModalWindow)
    in ModalWindow (created by MultiTabFormWithHeaderV3)
    in MultiTabFormWithHeaderV3 (created by CanvasElement)
    in CanvasElement (created by Canvas)
    in div (created by Canvas)
    in div (created by Canvas)
    in span
    in div
    in t (created by Canvas)
[... 53 lines cut ...]
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
app.js:966 Uncaught ReferenceError: elementData is not defined
    at MicroscopyMetadataTool.handleSaveMicroscope (app.js:966)
    at DropdownMenu.handleMenuItemClick (dropdownMenu.js:70)
    at Object.eval [as current] (DropdownItem.js:53)
    at eval (useEventCallback.js:11)
    at handleClick (SafeAnchor.js:45)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:327 Uncaught ReferenceError: elementData is not defined
    at MicroscopyMetadataTool.handleSaveMicroscope (app.js:966)
    at DropdownMenu.handleMenuItemClick (dropdownMenu.js:70)
    at Object.eval [as current] (DropdownItem.js:53)
    at eval (useEventCallback.js:11)
    at handleClick (SafeAnchor.js:45)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/ [... cut ...] dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.31.0-a1/31_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)

Akramhar, 2021-01-04:
should be solved in 0.32

strambc, 2021-01-05:
This is solved but issues remain as detailed in #154 and #155

closed 2021-01-05 by strambc