WU-BIMAC/MicroMetaApp-React #152: Uncaught fatal errors

strambc, 2021-01-05:
While fixing TESM microscope.
Saved to new Microscope file.
Saved multiple times.
Dragged multiple icons to new place.
Crashed While moving "FilterCube" and dragging EmissionFilter and Dichroic in new position

canvas.js:1131 Uncaught TypeError: Cannot read property 'Schema_ID' of null
    at eval (canvas.js:1131)
    at Array.forEach (<anonymous>)
    at Canvas.createList (canvas.js:1127)
    at Canvas.render (canvas.js:1599)
    at finishClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17160)
    at updateClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17110)
    at beginWork (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:18620)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527 The above error occurred in the <Canvas> component:
    in Canvas (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataToolContainer)
    in MicroscopyMetadataToolContainer (created by MicroscopyMetadataTool)
    in MicroscopyMetadataTool (created by MicroscopyMetadataToolComponent)
    in MicroscopyMetadataToolComponent

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
logCapturedError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:11102 Uncaught TypeError: Cannot read property 'Schema_ID' of null
    at eval (canvas.js:1131)
    at Array.forEach (<anonymous>)
    at Canvas.createList (canvas.js:1127)
    at Canvas.render (canvas.js:1599)
    at finishClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17160)
    at updateClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17110)
    at beginWork (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:18620)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
24bundle.js:6 Uncaught TypeError: Cannot read property 'style' of null
    at t.n.setCurrentTarget (bundle.js:6)
    at t.n.generateEnterLeaveEvents (bundle.js:6)
    at t.n.drag (bundle.js:6)
    at HTMLDocument.n.handleMouseMove (bundle.js:6)
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:88 Warning: Can't perform a React state update on an unmounted component. This is a no-op, but it indicates a memory leak in your application. To fix, cancel all subscriptions and asynchronous tasks in the componentWillUnmount method.
    in t (created by Canvas)
    in div (created by Canvas)
    in div (created by Canvas)
    in span
    in t (created by Canvas)
    in div (created by Canvas)
    in Canvas (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataToolContainer)
    in MicroscopyMetadataToolContainer (created by MicroscopyMetadataTool)
    in MicroscopyMetadataTool (created by MicroscopyMetadataToolComponent)
    in MicroscopyMetadataToolComponent
printWarning @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.32.0-a1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:88
25bundle.js:6 Uncaught TypeError: Cannot read property 'style' of null
    at t.n.setCurrentTarget (bundle.js:6)
    at t.n.generateEnterLeaveEvents (bundle.js:6)
    at t.n.drag (bundle.js:6)
    at HTMLDocument.n.handleMouseMove (bundle.js:6)

Akramhar, 2021-01-05:
I seem to be unable to replicate this issue

strambc, 2021-01-11:
Presented again.
Moved Optics Holder to Canvas
Added Lens
Attempted to move Optics Holder

strambc, 2021-01-11:
canvas.js:1131 Uncaught TypeError: Cannot read property 'Schema_ID' of null
    at eval (canvas.js:1131)
    at Array.forEach (<anonymous>)
    at Canvas.createList (canvas.js:1127)
    at Canvas.render (canvas.js:1617)
    at finishClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17160)
    at updateClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17110)
    at beginWork (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:18620)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527 The above error occurred in the <Canvas> component:
    in Canvas (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataToolContainer)
    in MicroscopyMetadataToolContainer (created by MicroscopyMetadataTool)
    in MicroscopyMetadataTool (created by MicroscopyMetadataToolComponent)
    in MicroscopyMetadataToolComponent

Consider adding an error boundary to your tree to customize error handling behavior.
Visit https://fb.me/react-error-boundaries to learn more about error boundaries.
logCapturedError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:19527
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:11102 Uncaught TypeError: Cannot read property 'Schema_ID' of null
    at eval (canvas.js:1131)
    at Array.forEach (<anonymous>)
    at Canvas.createList (canvas.js:1127)
    at Canvas.render (canvas.js:1617)
    at finishClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17160)
    at updateClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17110)
    at beginWork (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/alpha_0.33.4-a1/33.4_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:18620)
[... cut ...] ata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react/cjs/react.development.js:471
onWindowResize @ app.js:317
267bundle.js:6 Uncaught TypeError: Cannot read property 'style' of null
    at t.n.setCurrentTarget (bundle.js:6)
    at t.n.generateEnterLeaveEvents (bundle.js:6)
    at t.n.drag (bundle.js:6)
    at HTMLDocument.n.handleMouseMove (bundle.js:6)
n.setCurrentTarget @ bundle.js:6
n.generateEnterLeaveEvents @ bundle.js:6
n.drag @ bundle.js:6
n.handleMouseMove @ bundle.js:6

strambc, 2021-01-11:
Reproduced twice

closed 2021-01-12 by strambc