WU-BIMAC/MicroMetaApp-React #182: GUILBERT #1_config conflict between Filter cube turret and Em filter

ThomasGuilbert, 2021-02-09:
Hi,
Just an issue I got, on inverted stand, when filter cubes are asset with filters, once the Filter cube turret is in position too, I find no way to edit emission filters anymore. 
Whatever the zoom, the mouse position keep proposing to edit the Filter cube turret.
++

strambc, 2021-02-09:
I was able to reproduce this issue.
The FilterCubeTurrett icon has to be modified to allow access to Em-Filter

strambc, 2021-02-24:
[image]

Akramhar, 2021-02-27:
should be done in 0.35

strambc, 2021-03-15:
THis is not FIXED. The position of the SLOT is moved down but the snap in place remains at the same place as before

closed 2021-03-16 by strambc