WU-BIMAC/MicroMetaApp-React #193: Manage Settings --> Tool tips text has to be updated

strambc, 2021-02-24:
The tooltips for the following buttons have to be updated:

Edit Settings --> "Edit Settings: click this button to enter general information about the acquisition settings utilized for this image"

Validate @tier:X --> "Validation Tier selector: Choose the Tier level you want to use to validate this Settings file. etc."

Save settings --> "Choose whether you want to save this Settings file to the active Micro-Meta App repository or if you want to export it to your local file system."

strambc, 2021-02-24:
The tooltips for the following buttons have to be added:

Akramhar, 2021-03-08:
should be done in 0.36

strambc, 2021-03-14:
The tool tips the bottom row of buttons are not up to date.

Akramhar, 2021-03-15:
should be done in 0.37

closed 2021-03-16 by strambc