WU-BIMAC/MicroMetaApp-React #213: Manage Settings --> Available camera missing from Select Detector

strambc, 2021-02-24:
TESM microscope
Camera is available in Microscope File but missing from available options

strambc, 2021-02-24:
channelCanvas_V2.js:362 Uncaught TypeError: Cannot read property 'modelSettings' of undefined
    at ChannelCanvas_V2.onInnerElementDataSave (channelCanvas_V2.js:362)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287)
    at forEachAccumulated (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3259)
onInnerElementDataSave @ channelCanvas_V2.js:362
callCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188
invokeGuardedCallbackDev @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237
invokeGuardedCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292
invokeGuardedCallbackAndCatchFirstError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306
executeDispatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom [... cut ...] 1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:806
dispatchDiscreteEvent @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.34.0-b1-1/4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:4168

Akramhar, 2021-02-27:
should be fixed in 0.35

closed 2021-03-02 by strambc