WU-BIMAC/MicroMetaApp-React #222: Manage Settings: Change descriptions for Add multiple Planes interface

strambc, 2021-02-24:
1. General --> "This interface allows you to add multiple image Planes to this Image. For example, in case the Image has 10 Z-sections, 3 channels, and 1 time-points, the interface should be used to add three batches (one per Channel) of 10 Planes each."
2. NumberOfPlanes --> "Insert the number of Planes to be inserted as part of this batch".
3. ZIncrement --> "Select this if you want the different Planes in this batch to have increasing Z-dimension numbers."
4. TIncrement --> "Select this if you want the different Planes in this batch to have increasing timepoint-dimension numbers."
5. CIncrement --> "Select this if you want the different Planes in this batch to have increasing channel-dimension numbers."

Akramhar, 2021-02-27:
should be fixed in 0.35

closed 2021-03-02 by strambc

strambc, 2021-03-14:
This is related to #253