WU-BIMAC/MicroMetaApp-React #249: FATAL ERROR --> when selecting Objective

strambc, 2021-03-14:
TESM microscope
created Channel 01
Attempted to select an Objective

settingsMainView.js:233 Uncaught TypeError: Cannot read property 'Component_ID' of null
    at SettingMainView.onElementDataSave (settingsMainView.js:233)
    at SettingComponentSelector.onConfirm (settingComponentSelector.js:152)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287)
onElementDataSave @ settingsMainView.js:233
onConfirm @ settingComponentSelector.js:152
callCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188
invokeGuardedCallbackDev @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237
invokeGuardedCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292
invokeGuardedCallbackAndCatchFirstError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306
executeDispatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389
executeDispatchesInOrder @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414
executeDispatchesAndRelease @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278
executeDispatchesAndReleaseTopLevel @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287
forEachAccumulated @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3259
runEventsInBatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3304
runExtractedPluginEventsInBatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3514
handleTopLevel @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3558
batchedEventUpdates$1 @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/nod [... cut ...] copy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:806
dispatchDiscreteEvent @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:4168

Akramhar, 2021-03-15:
should be solved in 0.37

strambc, 2021-03-16:
This issue will be closed but it is related to a new issue that was just opened

closed 2021-03-16 by strambc