WU-BIMAC/MicroMetaApp-React #260: FATAL ERROR --> tried to open fswt-6hvirus-10minfix-stk_4-epi.tif.ome.tif.json

strambc, 2021-03-15:
TESM JSON file
When opening the saved settings file

----
app.js:1066 Uncaught TypeError: Cannot read property 'ValidationTier' of undefined
    at MicroscopyMetadataTool.createOrUseSettingFromSelectedFile (app.js:1066)
    at MicroscopyMetadataTool.createOrUseSetting (app.js:1151)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
    at invokeGuardedCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292)
    at invokeGuardedCallbackAndCatchFirstError (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306)
    at executeDispatch (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389)
    at executeDispatchesInOrder (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414)
    at executeDispatchesAndRelease (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278)
    at executeDispatchesAndReleaseTopLevel (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287)
createOrUseSettingFromSelectedFile @ app.js:1066
createOrUseSetting @ app.js:1151
callCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188
invokeGuardedCallbackDev @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237
invokeGuardedCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292
invokeGuardedCallbackAndCatchFirstError @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:306
executeDispatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:389
executeDispatchesInOrder @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:414
executeDispatchesAndRelease @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3278
executeDispatchesAndReleaseTopLevel @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3287
forEachAccumulated @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3259
runEventsInBatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3304
runExtractedPluginEventsInBatch @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3514
handleTopLevel @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:3558
batchedEventUpdates$1 @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs [... cut ...] copy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:806
dispatchDiscreteEvent @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_alpha_releases/current release/beta_0.36.0-b1-1/0.36.0-b1-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:4168

strambc, 2021-03-15:
[fswt-6hvirus-10minfix-stk_4-epi.tif.ome.tif.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/6138649/fswt-6hvirus-10minfix-stk_4-epi.tif.ome.tif.txt)
[STRAMBIO_tesm_022421.txt](https://github.com/WU-BIMAC/4DNMicroscopyMetadataToolReact/files/6138650/STRAMBIO_tesm_022421.txt)

Akramhar, 2021-03-16:
should be solved in 0.37

closed 2021-03-16 by strambc