WU-BIMAC/MicroMetaApp-React #278: Manage Settings --> Edit Tooltip text

strambc, 2021-03-16:
- [ ] Edit Planes --> "Click this button to enter or edit general information about the Image Planes."
- [ ] Edit Channels --> "Click this button to enter or edit general information about the image Channels and about the light path associated with each of them."
- [ ] Edit Microscope Settings --> "Click this button to enter or edit information about the Settings that were applied to the Microscope Stand during the acquisition of this Image."
- [ ] Edit Sample Positioning Settings --> "Click this button to select one or more Sample Positioning devices and edit the settings that were applied to them during the acquisition of this Image."
- [ ] Edit Microscope Table Settings --> "Click this button to select the Microscope Table and edit the settings that were used for the acquisition of this Image."
- [ ] Add multiple planes --> "Click this button to add multiple Planes to this Image."

Akramhar, 2021-03-16:
should be done in 0.39

strambc, 2021-03-16:
@Akramhar THREE MISSING ITEMS

Akramhar, 2021-03-17:
should be solved in 0.40

closed 2021-03-19 by strambc