WU-BIMAC/MicroMetaApp-React #286: FATAL ERROR --> Environmental, select, add, confirm select to edit settings CRASH

strambc, 2021-03-17:
/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react/cjs/react.development.js:315 Warning: Each child in a list should have a unique "key" prop.

Check the render method of `SettingComponentSelector`. See https://fb.me/react-warning-keys for more information.
    in div (created by SettingComponentSelector)
    in SettingComponentSelector (created by SettingMainView)
    in SettingMainView (created by MicroscopyMetadataTool)
    in div (created by MicroscopyMetadataToolContainer)
    in MicroscopyMetadataToolContainer (created by MicroscopyMetadataTool)
    in MicroscopyMetadataTool (created by MicroscopyMetadataToolComponent)
    in MicroscopyMetadataToolComponent
printWarning @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react/cjs/react.development.js:315
settingComponentSelector.js:648 Uncaught TypeError: Cannot read property 'ID' of null
    at eval (settingComponentSelector.js:648)
    at Array.forEach (<anonymous>)
    at eval (settingComponentSelector.js:646)
    at Array.forEach (<anonymous>)
    at SettingComponentSelector.render (settingComponentSelector.js:635)
    at finishClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17160)
    at updateClassComponent (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17110)
    at beginWork (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:18620)
    at HTMLUnknownElement.callCallback (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188)
    at Object.invokeGuardedCallbackDev (/Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237)
eval @ settingComponentSelector.js:648
eval @ settingComponentSelector.js:646
render @ settingComponentSelector.js:635
finishClassComponent @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17160
updateClassComponent @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:17110
beginWork @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:18620
callCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:188
invokeGuardedCallbackDev @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:237
invokeGuardedCallback @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:292
beginWork$1 @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:23203
performUnitOfWork @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:22157
workLoopSync @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:22130
performSyncWorkOnRoot @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:21756
(anonymous) @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:11089
unstable_runWithPriority @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/scheduler/cjs/scheduler.development.js:653
runWithPriority$1 @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:11039
flushSyncCallbackQueueImpl @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.de [... cut ...] 9.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:806
dispatchDiscreteEvent @ /Users/strambc/Dropbox/Workstuff/4DN_work_2018_dropbox/2018-4DN_metadata standard + software_model development/4DN-OME_Micro-Meta-App_devel/Micro-Meta App Software/react_beta_releases/beta_0.39.1-b1-1/39-1_4dn-microscopy-metadata-tool-standalone.app/Contents/Resources/app.asar/node_modules/react-dom/cjs/react-dom.development.js:4168

strambc, 2021-03-17:
For Sample Positioning it happened when I attempted to click back into the SamplePositioningSettings after Adding Elements and Confirming

Akramhar, 2021-03-17:
should be done in 0.40

closed 2021-03-19 by strambc