WU-BIMAC/MicroMetaApp-React #290: [BUG] MANAGE SETTINGS --> Cannot delete Additional Elements after addition

strambc, 2021-03-17:
**Describe the bug**
After adding elements to the light path using "Add additional elements" it is not possible to delete them.
If one attempts to delete, confirms and gets back in, the items are again displayed and duplicated

**To Reproduce**
Steps to reproduce the behavior:
1. Go to: LightPath window
2. Click on 'Add additional element(s)'
3. Add elements.
4. Hit confirm
5. Click on the same button as before
6. Delete elements
7. Delete
8. Confirm
9. Confirm CHannel
10. Confirm Channels
11. Click on Edit Channels
12. Enter into the Channel above.
13. Click on the same 'Add additional element(s)'

**Expected behavior**
ELEMENTS ARE BACK AND SOME ARE DUPLICATED

**Screenshots**

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version: beta v40-1
 - Stand Alone

strambc, 2021-03-17:
https://user-images.githubusercontent.com/4050585/111545818-0e796380-874d-11eb-84ba-e175d5d03cd1.mov

Akramhar, 2021-03-21:
should be solved in 0.41

closed 2021-03-22 by strambc