WU-BIMAC/MicroMetaApp-React #294: ALERT MODEL CHANGE --> Changed Advanced and Confocal/ConfocalScannerSettings/EffectiveZoom description

strambc, 2021-03-18:
**Report an approved change to existing metadata model structure, elements, fields names, descriptions, data types, cardinality, requirement level, and/or enumeration values.**

**Current Element name:** ConfocalScannerSettings <br/>

**Current Field name:** EffectiveZoom <br/>

**Source of change (e.g., #XXX issue with change request):** INTERNAL WITH SUPPORT FROM ORESTIS FAKLARIS <br/>

**Brief description of the change and its rationale (short and compelling):** After consulting with Oresti Faklaris, it was decided to expand the description for this field to make it more clear to users. <br/>

Current Field description: "This field describes the Effective Zoom factor that was utilized for the acquisition of this Image (e.g., zoom = 1.3). "<br/>

**New Field description:** "This field describes the Effective Zoom factor that was utilized for the acquisition of this Image (e.g., zoom = 1.3). In confocal microscopy, it is possible to set a given hardware-based Zoom factor, aka "Confocal Zoom" or "Scan Zoom", by reducing the angle at which the scanning mirrors operate. As a result, the scanned area will be reduced in size, which (because the pixel areas associated with the scan are kept constant) will decrease the effective size of the portion of the sample that is projected on each pixel thus increasing the effective resolving power of the system. For example, with a 512x512 pixel area, and with a 40x objective, zoom 1 we result in a pixel size of 732nm, while zoom 10 will reduce the pixel size to 73nm).In confocal microscopy, it is possible to set a given hardware-based Zoom factor, aka "Confocal Zoom" or "Scan Zoom", by reducing the angle at which the scanning mirrors operate. As a result, the scanned area will be reduced in size, which (because the pixel areas associated with the scan are kept constant) will decrease the effective size of the portion of the sample that is projected on each pixel thus increasing the effective resolving power of the system. For example, with a 512x512 pixel area, and with a 40x objective, zoom 1 we result in a pixel size of 732nm, while zoom 10 will reduce the pixel size to 73nm)."

Akramhar, 2021-03-21:
should be done in 0.41

Akramhar, 2021-03-23:
should be done in 0.42.1

closed 2022-03-30 by strambc