WU-BIMAC/MicroMetaApp-React #301: [BUG] MANAGE SETTINGS --> Add additional elements does not allow to add when element is not validated for a given Tier

strambc, 2021-03-19:
**Describe the bug**
When a given element is not validated for a given Tier, Add Additional elements does not allow to add it. 
Instead, the element should be allowed to be added but maintain a red dot

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Edit Channels, Edit a given Channel
2. Click on Add Additional Elements
3. Attempt to add an element that is not validated at the present tier
4. Element is not added

**Expected behavior**
The element should be added but maintain a red dot to alert the user that the element is not validated and should be modified in Manage Instrument

**Desktop (please complete the following information):**
 - OS: Mac
 - Version [e.g. 22]: v040

strambc, 2021-03-22:
checked again with v41-2

To test added NewCouplingLens to TESM and left invalidated on purpose at Tier 3
When attempting to add CouplingLens it appears in the list but cannot be added.
Is this the desired behavior now?

Akramhar, 2021-03-23:
chosen solution: display validation dot on component that are incomplete and dont allow them to be "add" or "confirm"

should be done in 0.42

closed 2021-03-23 by strambc