WU-BIMAC/MicroMetaApp-React #303: [BUG] MANAGE INSTRUMENT/SETTINGS --> SamplePositioningSettings added to all added elements

strambc, 2021-03-19:
**Describe the bug**
When entering Attribute value in SamplePositioningSettings is added to all added elements

**To Reproduce**
Steps to reproduce the behavior:
1. Go to SamplePositioningSettings
2. Add multiple elements
3. Enter a value in one of the Settings
4. Same value appears in the same position in all elements

**Expected behavior**
Each element should have different attribute values

https://user-images.githubusercontent.com/4050585/111755920-0534e780-8870-11eb-91dc-bb5a6381be6f.mov

**Desktop (please complete the following information):**
 - OS: Mac
 - Version [e.g. 22]: v040

strambc, 2021-03-19:
Only affects elements that have already been added

Akramhar, 2021-03-21:
should be solved in 0.41

closed 2021-03-22 by strambc